revk/SolarSystem #28: Option to use this code to monitor galaxy system?

NLRene, 2019-01-19:
More of a question/request to author instead of issue. I have an existing Honeywell Galaxy system (G2-44), and I was hoping to be able to use this code to capture the status of the various sensors (PIR, etc) for my home automation system by monitoring the RS485 bus with an rpi and this code.

So far I'm able to get consistent HEX output using the galaxybus.c code, but I struggle to get make any sense of it / trace it back to any devices/sensors/etc. Any hints/thoughts? Possible at all?

Thx!

revk, 2019-01-20:
I don't think they are well documented. The code does handle several Galaxy devices (keypad, Max reader, and I/O units) which we have mostly worked out. If you have other things, it would need some trial and error.

closed 2019-01-20 by revk

NLRene, 2019-01-26:
Dear RevK, Adrian,

Thank you for your response. I hope you can also help me out with just
understanding how I can use the code to read RIO and keypad data. Let me
explain my setup first:

- I have a running Honeywell Galaxy G2-44+
- I have a rPi3 running your code, using an USB RS485
- I have connected the USB with the RS 485 bus of the galaxy
- I have adapted your code a bit: not writing to the bus, only reading:

   - Use galaxybus.c / galaxubus.h only
   - Have adapted code a bit, using calling in the main function only poller
   - Outcommented function sendcmd to avoid sending commands
   - With debug on I get the following dump, which shows some repeating hex
   output, but I struggle to interpret this feed
   - Can you give me some hints on how to interpret the received bytes?

I've attached a dump of the feedback received.

Thx again!

Rene.

pi@raspberrypi:~/Scripts/Galaxy $ ./galaxybus.o
starting poller, busid 1
galaxybus.o: Cannot open /sys/bus/usb-serial/devices/ttyUSB1/latency_timer: Permission denied
XXX200 < 6D B9 2D 77 42 FF FF FF FF FF A9 D7 FD 57 77 80 11 BF C3 69 77 01 02 23 4F 9F 00 77 80 8B 6D B9 2D 77 (0.0ms)
Bad checksum
XXX200 < 6D B9 (0.0ms)
XXX200 < 77 42 FF FF FF FF FF (0.0ms)
XXX200 < D7 FD (0.0ms)
XXX200 < 77 80 (0.0ms)
XXX200 < BF C3 (0.0ms)
XXX200 < 77 01 01 (0.0ms)
XXX200 < 4F 9F (0.0ms)
XXX200 < 77 80 (0.0ms)
XXX200 < 6D B9 (0.0ms)
XXX200 < 77 42 FF FF FF FF FF (0.0ms)
XXX200 < BF C3 (0.0ms)
XXX200 < 77 01 02 (0.0ms)
XXX200 < 4F 9F (0.0ms)
XXX200 < 77 (0.0ms)
XXX200 < FF FF FF FF FF FF FF FF FF FF FF (0.0ms)
PAD200 found new
PAD200 lost
PAD200 < 6D B9 (0.0ms)
PAD200 < 77 42 FF FF FF FF FF (0.0ms)
PAD200 < D7 FD (0.0ms)
PAD200 < 77 80 (0.0ms)
PAD200 < BF C3 69 77 01 02 (0.0ms)
PAD200 < 4F 9F (0.0ms)
PAD200 < 77 80 (0.0ms)
PAD200 < 6D B9 (0.0ms)
PAD200 < 77 42 FF FF FF FF FF (0.0ms)
PAD200 < BF E3 FF F3 F3 ED E7 E1 23 4D (0.0ms)
PAD200 < 77 00 B7 6A F2 (0.0ms)
PAD200 < 4F 9F 00 77 80 (0.0ms)
PAD200 < BF E3 FF F3 F3 ED E7 E1 23 4D (0.0ms)
PAD200 < 77 00 B7 6A 79 (10.8ms)
PAD200 < 6D B9 (10.2ms)
PAD200 < 77 42 FF FF FF FF FF (0.0ms)
PAD200 < D7 FD (0.0ms)
PAD200 < 77 80 (1.7ms)
PAD200 < BF C3 (0.1ms)
PAD200 < 77 07 7E 07 7E 07 7E 07 7E 07 7E 07 7E 07 7E 07 7E 07 D9 C0 FF FE E7 (8.7ms)
PAD200 < 4F 9F 00 77 80 (9.6ms)
PAD200 < 6D B9 (0.0ms)
PAD200 < 77 42 FF FF FF FF FF (0.2ms)
PAD200 < D7 FD (2.7ms)
[... 386 lines cut ...]
PAD200 < D7 FD (0.1ms)
PAD200 < 77 80 (0.0ms)
PAD200 < BF C3 (0.1ms)
PAD200 < 77 01 01 (0.0ms)
PAD200 < 4F 9F (0.1ms)
PAD200 < 77 80 (0.0ms)
PAD200 < 6D B9 (0.0ms)
PAD200 < 77 42 FF FF FF FF FF (0.0ms)
PAD200 < BF C3 (3.4ms)
PAD200 < 77 01 02 (0.0ms)
PAD200 < 4F 9F (0.0ms)
PAD200 < 77 80 (0.9ms)
PAD200 < 6D B9 (0.0ms)
PAD200 < 77 42 FF FF FF FF FF (0.3ms)
PAD200 < D7 FD (0.0ms)
PAD200 < 77 80 (0.0ms)
PAD200 < BF C3 (0.0ms)
PAD200 < 77 01 01 (0.0ms)
PAD200 < 4F 9F (0.0ms)
PAD200 < 77 80 (0.0ms)
PAD200 < 6D B9 (0.0ms)
PAD200 < 77 42 FF FF FF FF FF (0.0ms)
PAD200 < BF C3 (5.4ms)
PAD200 < 77 07 7E 07 7E 07 7E 07 7E 87 76 7E 87 76 7E 87 99 88 FF FE (0.0ms)
^C

revk, 2019-01-27:
You will need to do more than simply not sending data. There is polling and so each response is a response to a poll. To passively monitor you need code that understands the poll and then the response to that poll. It would also have to work out, or know. What type of device is at each address.

I did write up quite a bit on my blog I believe. That may help.