revk/SolarSystem #31: Installation guide

OptimusGREEN, 2021-03-29:
Hi, 

This looks like a great project so thanks for creating it.

I am struggling to find the installation guide within the manual pages though. I see you can install it on a pi and then connect the pi to the rs485 bus which is great.

Could you point me to the install guide for installing the software on my pi please?

revk, 2021-03-29:
Hmm, I'll have to look. It used off the shelf USB->RS485 widgets, and I thought there were even RS part numbers in there. The code just needs making and running. How far did you get?

OptimusGREEN, 2021-03-29:
yeah its just the software build i couldnt find info for.

`john@MacBook SolarSystem % make
git submodule update --init
cc -g -Wall -Wextra -O -c -o galaxybus.o galaxybus.c -I. -IAXL -DLIB -pthread
galaxybus.c:28:10: fatal error: 'popt.h' file not found
#include <popt.h>
         ^~~~~~~~
1 error generated.
make: *** [galaxybus.o] Error 1`

make isn't something im familiar with.

revk, 2021-03-29:
Ah well install popt for a start!

OptimusGREEN, 2021-03-29:
yup, i've installed each requirement as it errored but now having an issue with mqtt being undeclared.

`git submodule update --init
cc -g -Wall -Wextra -O -o alarmpanel alarmpanel.c galaxybus.o port.o afile.o door.o -I. -IAXL -IDataformat -Iwebsocket -IDESFireAES/include AXL/axl.o Dataformat/dataformat.o websocket/websocketxml.o DESFireAES/desfireaes.o -lcurl -pthread -lpopt -DLIBWS   -lcrypto -lssl
alarmpanel.c: In function ‘mqtt_door’:
alarmpanel.c:382:10: warning: implicit declaration of function ‘mosquitto_publish’ [-Wimplicit-function-declaration]
          mosquitto_publish(mqtt, NULL, topic, afilelen, afile, 1, 0);
          ^~~~~~~~~~~~~~~~~
alarmpanel.c:382:28: error: ‘mqtt’ undeclared (first use in this function)
          mosquitto_publish(mqtt, NULL, topic, afilelen, afile, 1, 0);
                            ^~~~
alarmpanel.c:382:28: note: each undeclared identifier is reported only once for each function it appears in
alarmpanel.c: In function ‘mqtt_led’:
alarmpanel.c:401:22: error: ‘mqtt’ undeclared (first use in this function)
    mosquitto_publish(mqtt, NULL, topic, strlen(seq), seq, 1, 0);
                      ^~~~
alarmpanel.c: In function ‘mqtt_output’:
alarmpanel.c:417:22: error: ‘mqtt’ undeclared (first use in this function)
    mosquitto_publish(mqtt, NULL, topic, 1, &msg, 1, 0);
                      ^~~~
alarmpanel.c: In function ‘keypad_send’:
alarmpanel.c:766:25: error: ‘mqtt’ undeclared (first use in this function)
       mosquitto_publish(mqtt, NULL, topic, 32, message, 1, 0);
                         ^~~~
alarmpanel.c: In function ‘doevent’:
alarmpanel.c:3664:34: error: ‘mqtt’ undeclared (first use in this function)
                mosquitto_publish(mqtt, NULL, topic, afilelen, afile, 1, 0);
                                  ^~~~
make: *** [Makefile:33: alarmpanel] Error 1`

im just running make from the main solar system directory. Are there other steps required to build? 
Sorry, this stuff is kind of new, i know some python but thats no help here lol.

OptimusGREEN, 2021-03-29:
ok, i had the wrong mosquitto lib installed and have got passed that bit.

I'm now stuck with what only seems like warnings to me rather than the build erroring out but I don't have a build so I guess I'm wrong.

`git submodule update --init
Submodule path 'DESFireAES': checked out 'b61d46c4ac81d042bb7dc18f8539fde37a1c596b'
make -C DESFireAES
make[1]: Entering directory '/media/psf/Home/Development/git/SolarSystem/DESFireAES'
cc -fPIC -O -DLIB -c -o desfireaes.o -Iinclude desfireaes.c
make[1]: Leaving directory '/media/psf/Home/Development/git/SolarSystem/DESFireAES'
cc -g -Wall -Wextra -O -o alarmpanel alarmpanel.c galaxybus.o port.o afile.o door.o -I. -IAXL -IDataformat -Iwebsocket -IDESFireAES/include AXL/axl.o Dataformat/dataformat.o websocket/websocketxml.o DESFireAES/desfireaes.o -lcurl -pthread -lpopt -DLIBWS  -DLIBMQTT -lmosquitto -lcrypto -lssl
alarmpanel.c: In function ‘do_keypad_update’:
alarmpanel.c:2992:27: warning: ‘%02d’ directive output may be truncated writing between 2 and 11 bytes into a region of size between 5 and 12 [-Wformat-truncation=]
    snprintf(l2, 17, "%04d-%02d-%02d %02d:%02d", lnow.tm_year + 1900, lnow.tm_mon + 1, lnow.tm_mday, lnow.tm_hour, lnow.tm_min);
                           ^~~~
alarmpanel.c:2992:21: note: directive argument in the range [-2147483647, 2147483647]
    snprintf(l2, 17, "%04d-%02d-%02d %02d:%02d", lnow.tm_year + 1900, lnow.tm_mon + 1, lnow.tm_mday, lnow.tm_hour, lnow.tm_min);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
alarmpanel.c:2992:4: note: ‘snprintf’ output between 17 and 60 bytes into a destination of size 17
    snprintf(l2, 17, "%04d-%02d-%02d %02d:%02d", lnow.tm_year + 1900, lnow.tm_mon + 1, lnow.tm_mday, lnow.tm_hour, lnow.tm_min);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
alarmpanel.c:2805:42: warning: ‘%2d’ directive output may be truncated writing between 2 and 10 bytes into a region of size 3 [-Wformat-truncation=]
          snprintf(l1, 17, "SETTING ALARM %2d", left);
                                          ^~~
alarmpanel.c:2805:27: note: directive argument in the range [0, 2147483647]
          snprintf(l1, 17, "SETTING ALARM %2d", left);
                           ^~~~~~~~~~~~~~~~~~~
alarmpanel.c:2805:10: note: ‘snprintf’ output between 17 and 25 bytes into a destination of size 17
          snprintf(l1, 17, "SETTING ALARM %2d", left);
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
alarmpanel.c: In function ‘doevent’:
alarmpanel.c:3549:51: warning: ‘%02u’ directive output may be truncated writing between 2 and 10 bytes into a region of size 4 [-Wformat-truncation=]
             snprintf(doorno, sizeof(doorno), "DOOR%02u", d);
                                                   ^~~~
alarmpanel.c:3549:46: note: directive argument in the range [0, 2147483647]
             snprintf(doorno, sizeof(doorno), "DOOR%02u", d);
                                              ^~~~~~~~~~
alarmpanel.c:3549:13: note: ‘snprintf’ output between 7 and 15 bytes into a destination of size 8
             snprintf(doorno, sizeof(doorno), "DOOR%02u", d);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
alarmpanel.c:3304:51: warning: ‘%02u’ directive output may be truncated writing between 2 and 10 bytes into a region of size 4 [-Wformat-truncation=]
             snprintf(doorno, sizeof(doorno), "DOOR%02u", d);
                                                   ^~~~
alarmpanel.c:3304:46: note: directive argument in the range [0, 2147483647]
             snprintf(doorno, sizeof(doorno), "DOOR%02u", d);
                                              ^~~~~~~~~~
alarmpanel.c:3304:13: note: ‘snprintf’ output between 7 and 15 bytes into a destination of size 8
             snprintf(doorno, sizeof(doorno), "DOOR%02u", d);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc -g -Wall -Wextra -O -o cardissue cardissue.c -I. -IDESFireAES/include DESFireAES/desfireaes.o -IAXL AXL/axl.o -lcrypto -lpopt -pthread -lcurl -lmosquitto afile.o
cardissue.c: In function ‘recvt’:
cardissue.c:334:29: warning: passing argument 2 to restrict-qualified parameter aliases with argument 4 [-Wrestrict]
     int r = select (fd + 1, &fset, NULL, &fset, &t);
`

Any help would be greatly appreciated.

revk, 2021-03-30:
You can probably ignore the warnings. I'll have to look at them sometime as compilers have got a lot more fussy these days.

OptimusGREEN, 2021-04-03:
Ok so I've managed to get it built and installed on my pi using the following deps command

`sudo apt install git libpopt-dev libxmlsec1-dev libcurl4-openssl-dev libmosquitto-dev mosquitto mosquitto-clients mosquitto-dev`

I'm having issues getting it to tie in with my existing system without it saying reset tamper etc but I think this is mostly because I don't understand how to setup my xml file properly for my components.

My existing setup:

- Galaxy g2-20 panel
- RF Portal (Address 5)
- 3 x RF PIR
- 1 x RF reed switch
- CP038 keypad with built in prox reader (Address 0)

The existing keypad is connected to my panel using the A:B lines.

My pi is connected to my panel using the A:B lines via a max485 usb adapter.

This is what i have in my config. I know PAD110 is my physical pad, I have also tried with a virtual pad at 120. 

`<?xml version="1.0" encoding="UTF-8"?>
<panel>
   <system name="SolarSystem" ws-port="80" engineering="*" mqtt-host="ssmqtt.iot" mqtt-user="solarsystem" mqtt-pass="**********" mqtt-ca="/etc/mosquitto/ca.crt" iot-$
   <group id="0" name="TeamGREEN"/>
   <keypad id="PAD110" message="KeyPad" name="Kpad" prox="190" silent="true" groups="*"/>
   <user name="john" pin="1234" pass="test" full-name="John Green" reset="*" arm="*" disarm="*" rearm="*"/>
</panel>`

If you had my setup, what would your config file look like?

I have also pasted my log [here](https://pastebin.com/raw/hczjTgzy) case there is any info in there that might trigger something. It just keeps saying there is a tamper on my keypad but when I remove my pi/SS and reset my alarm it is all normal.

Any detailed help would be greatly appreciated.

selfmon-automation, 2021-04-10:
> My existing setup:
> 
> * Galaxy g2-20 panel
> * RF Portal (Address 5)
> * 3 x RF PIR
> * 1 x RF reed switch
> * CP038 keypad with built in prox reader (Address 0)
> 
> The existing keypad is connected to my panel using the A:B lines.
> 
> My pi is connected to my panel using the A:B lines via a max485 usb adapter.
> 

@OptimusGREEN the SolarSystem is supposed to replace your G2-20 panel, so you need to disconnect the panel and replace it with something like a power RIO.

While the SolarSystem project adds additional features and workarounds for some of the Galaxy settings that annoyed @revk with his own Galaxy system, it does lack some features that the standard Galaxy panels provide in their efforts to cover many installation scenarios and as part of their standards compliance for insurance approved systems.

OptimusGREEN, 2021-04-11:
AHH, I didn't realise this. I thought it was in addition to the panel to allow mqtt communication.

I appreciated you taking the time to give me this info. 

I'll close this now as it's obviously not the project for me.

closed 2021-04-11 by OptimusGREEN

selfmon-automation, 2021-04-11:
Take a look at the SelfMon vmod.  It adds mqqt and will provide RF device payloads along with virtualised keypad and more. Unfortunately, the vmod module is a product and not an open source project.