revk/SolarSystem #67: mbedTLS memory leak

revk, 2021-07-15:
mbedTLS has huge memory leak for various types of non clean close.

Causes all sorts of issues.

ghenry, 2021-07-15:
What's the reason for using mbedTLS vs say WolfSSL or OpenSSL? Size?

revk, 2021-07-15:
Just what came with ESP IDF to be honest.

Is WolfSSL better?

ghenry, 2021-07-15:
Ah, OK. I'm really enjoying WolfSSL. Great examples - https://github.com/wolfSSL/wolfssl/tree/master/examples and lovely docs - https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/README.md

revk, 2021-07-16:
After much looking, seems all I need is :-

`git submodule add --recursive https://github.com/espressif/esp-wolfssl.git components/esp-wolfssl`

This provides the options in `make menuconfig`, yay!

Sadly, that does not actually work. It seems not to actually build the library!

`make: *** No rule to make target '/projects/github/SolarSystem/ESP32/components/esp-wolfssl/wolfssl/lib/esp32/libwolfssl.a', needed by '/projects/github/SolarSystem/ESP32/build/SS.elf'.  Stop.`

And help welcome.

ghenry, 2021-07-16:
No sure. See what they say
https://github.com/espressif/esp-wolfssl/issues/10

ghenry, 2021-07-16:
I'm just following the docs myself to see if I can get it build. The ESPs are nice. Soooo many options in menuconfig. Just like buidling the Linux kernel.

revk, 2021-07-16:
Apparently handling (legacy) `make` command was not in there and will be soon.

revk, 2021-07-19:
Needs an extra component.mk to fix for Make for now.

But works.

closed 2021-07-19 by revk