revk/SolarSystem #74: Inconsistent armok/disarmok

revk, 2021-07-16:
We say armok if arming is allowed for the areas we arm.
We say disarmok if disarming is allowed for areas we disarm, and those are armed.

We should do one or the other really. Or an attribute to say we did the arm/disarm.

Maybe we do like we do like elsewhere as a third state, `"armok":null` meaning we are allowed to arm but there is nothing to arm...

ghenry, 2021-07-16:
armstat:

armok
disarmok
null

Something like that or https://vanderlee.github.io/tristate/

Just an enum, no?

revk, 2021-07-16:
They are separate flags, and currently boolean. A fob can be Ok to arm, and Ok to disarm. It is mainly down to the settings in the fob.

I could just report based on fob setting, or report on whether there is any effect, i.e. we are armed, so armok false.

I am thinking armok:true means we are allowed to arm, amok:false means we are not allowed to arm, and amok:null means it is a moot point as we are not armed for what the fob is allowed to do. Or some such.

It's all JSON based, and would be used by a control system decided what to do if the door is not in an autonomous mode.

ghenry, 2021-07-16:
OK, understood.

Is it not better to report what state it is in?

armed: true
armed: false

Thanks.

revk, 2021-07-16:
Well armed is slightly more complex and reported in a different way.
But I think the best one is not send armok if armed. Same with disarmok... No confusion then.

ghenry, 2021-07-16:
That makes more sense and doesn't hurt my head.

revk, 2021-07-17:
OK, I am going to write up the logic and implement it separately - this ends up as part of a larger clarification of the operation of fobs at doors.

closed 2021-07-17 by revk