revk/SolarSystem #82: On site live view

revk, 2021-07-18:
Can javascript talk mqtt directly yet? If so, that would be great - the whole thing could be in javascript maybe, talking to IoT MQTT server.

Of course with suitable vpn, off-site live view as well.

ghenry, 2021-07-18:
Looks like it https://github.com/mqttjs/MQTT.js/

There seems to be loads of choice. Thus one has lots of stars but also LOTS of open issues.

revk, 2021-07-18:
F

> Looks like it https://github.com/mqttjs/MQTT.js/
> 
> There seems to be loads of choice. Thus one has lots of stars but also LOTS of open issues.

Fun!

ghenry, 2021-07-18:
A bit older and less stars https://www.eclipse.org/paho/index.php?page=clients/js/index.php
https://github.com/eclipse/paho.mqtt.javascript

Usual GitHub due diligence needed:

- Stars
- Open issues
- Closed issues
- Age of last commit 
- Docs 
- Use by other projects 
- Bloat 
- npm options 

etc. 

Navigating the javascript project world is paralysing so pick one and move on :)

revk, 2021-07-18:
One option would be the one I often go for - code my own :-)
I'll take a look though.

ghenry, 2021-07-18:
Yeah, I'm at the point where I'm sick of 3rd party libs and frameworks and they just die off. Standards based things are very attractive.

This is a great listen about that https://corecursive.com/066-sqlite-with-richard-hipp/

revk, 2021-09-20:
I am think this is a separate major project - maybe even a separate GitHub project, once the alarm system itself is stable. So closing for now.

closed 2021-09-20 by revk