revk/SolarSystem #88: Review adopting fobs UI

Serenity-Whitefall, 2021-07-18:
When adopting a fob the first state shown in the notes field after clicking adopt is "Default AID () waiting to be adopted". I.e. linked to the AID "Default AID" and with an empty access class. 

If you then go and edit the fob to add a name but don't select any access (default is No access) that results in the notes field updating to "No AIDs".

If you don't edit the fob and place on the NFC reader the notes field changes to "Default AID ()".

In either case the fob is not usable as it doesn't have an access class assigned to it. Would it make more sense that when adopting a fob you have to select an access class for it? I can see a use case for wanting to adopt a fob to the organisation but not grant access right away. Could that be just setup with No AIDs? Is there a use case for a fob with a AID assigned but no access class?

revk, 2021-07-19:
You can select an AID, that is an option when provisioning...

revk, 2021-07-22:
I propose changing this, and making interactive only.

i.e. administrator has a reader on their desk, or is accessing control pages from a door (e.g. on their phone).

Functions to :-

1. Identify fob - waits interactively for fob to be applied and then goes to fob edit page with AIDs and access
2. On the fob edit page interactive options to "adopt" and maybe "format"

This would be much easier to use.

commit 144a4aa1 (2021-07-22): Closes #88
closed 2021-07-22 by revk