revk/SolarSystem #93: Less clutter to cloud MQTT

revk, 2021-07-19:
Currently the RevK library handles connections to multiple MQTT servers. This is used for secure MQTT to cloud and local IoT MQTT.

The library has _copy functions allow messages to the main MQTT connection and copies to addition connections. These are used with various hot settings to copy messages to local IoT server.

However, this means a lot of clutter to main cloud server which is unnecessary - e.g. every input or output change or door state changed.

- Change library to allow sending to a bit map of servers
- Change MQTT mesh relay to cater for bit map
- Handle "none" as option, simply discarding message
- Change all calls to send to selected servers - only selected necessary messages to cloud server
- Add setting to send all messages to cloud for debug
- Add cloud control of debug setting (probably site wide)

revk, 2021-07-19:
Done

closed 2021-07-19 by revk

ghenry, 2021-07-19:
I'm sure you're aware, but in your git commit message, if you do Closes #93 then this issue gets closed when you git push. It's a nice way to tie up your work with the issue. Or Fixes #95

https://github.blog/2013-01-22-closing-issues-via-commit-messages/

Thanks.

revk, 2021-07-19:
Thank, I a far from an expert in git.