revk/SolarSystem #178: How to compile this ESP project ?

Noschvie, 2021-10-15:
Hello
is there a "HowTo" compiling this project ?
Using ESP-IDF 4.3 under Windows gets an Error "Failed to resolve component 'ESP32-RevK'."
thanks
Norbert

[image]

revk, 2021-10-15:
No idea on windows, did you recursively fetch everything?

Noschvie, 2021-10-21:
It seems that perl is needed, isn't it?

```
...
-- Found PythonInterp: C:/Users/nosch/.espressif/python_env/idf4.3_py3.8_env/Scr
ipts/python.exe (found version "3.8.7")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "SS" version: V1.0448
...
```

Noschvie, 2021-11-01:
Now I changed my development platform to Linux / Debian 11.1 running as VM under Hyper-V and get the project to be compiled and built.

My question : what have to be changed to define / config my local wifi ( SSID and password) ?

thanks, Norbert

revk, 2021-11-01:
make menuconfig
Allows all the defaults to be set including SSID, etc.

Noschvie, 2021-11-01:
> 
> 
> make menuconfig
Allows all the defaults to be set including SSID, etc.

... recompiled and connected to the local wifi. Perfect! Thanks

closed 2021-11-01 by Noschvie