revk/SolarSystem #183: Authenticated bug

Serenity-Whitefall, 2022-01-16:
I spotted some bugs whilst coding the API enhancement...

https://github.com/revk/SolarSystem/blob/b17ef467bc8be78533f6f216191b6a4629d5daec/www/provisiondevice.cgi#L59

It seems this always says "Device is authenticated." regardless of whether the database value is true or false. Should it actually be "if authenticated=true" ?

Also I'm not sure it can even be set to true in the database.

https://github.com/revk/SolarSystem/blob/b17ef467bc8be78533f6f216191b6a4629d5daec/solarsystem.c#L1316

For it to be set to true, secureid would have to evaluate to true in line 1316, however if it did evaluate to true, then it would have entered the if section at 1302 and therefore never reach the else section at 1310.

revk, 2022-01-16:
It looks wrong on both counts, indeed.

revk, 2022-01-16:
Ok I am not sure what I was intending for that field. Clearly the display can easily be fixed with the =true. But the setting does look wrong, and probably should be commented out for now. Ultimately if never set we should remove the field.

revk, 2022-01-16:
Commented out

commit 1375c486 (2022-01-16): Unsure of correct fix #183
closed 2022-01-16 by revk