revk/SolarSystem #184: Memory leak?

Serenity-Whitefall, 2022-01-16:
https://github.com/revk/SolarSystem/blob/b17ef467bc8be78533f6f216191b6a4629d5daec/solarsystem.c#L992-L1003

I think this sql_free_result at line 1000 doesn't want to be inside the else.

Also I'm not sure of the purpose of line 994 "deviceid = v". It gets overwritten in line 998 and if there wasn't a result returned then it would return from the function at line 1002 anyway?

revk, 2022-01-16:
I agree on the free, that looks like a leek. Setting deviceid may be used later anyway as it has wider scope even after the return.

closed 2022-01-16 by revk

ghenry, 2022-01-16:
Since thus us an open source project you should put in. Coverity auto scan. Very helpful. See here https://github.com/SentryPeer/SentryPeer/actions/runs/1704107864/workflow

revk, 2022-01-16:
No idea how you do that...

ghenry, 2022-01-16:
Sign up:

https://scan.coverity.com/users/sign_up

Add your project.

Get your key from the bottom of that page.

Add it to https://github.com/revk/SolarSystem/settings/secrets/actions like
attached.

then create this file in a .github/worksflow folder in your repo:

https://github.com/SentryPeer/SentryPeer/blob/main/.github/workflows/coverity-scan.yml

Change these lines:

https://github.com/SentryPeer/SentryPeer/blob/main/.github/workflows/coverity-scan.yml#L57
https://github.com/SentryPeer/SentryPeer/blob/main/.github/workflows/coverity-scan.yml#L36
https://github.com/SentryPeer/SentryPeer/blob/main/.github/workflows/coverity-scan.yml#L64

etc.

That's it.

Want me to do a fork and PR for you? You do the sign up bit and get your
Project Token from the Settings page in Coverity.

Thanks.

revk, 2022-01-16:
Yeh, a PR would be easiest for me as I have never done that! Give me a mo.

revk, 2022-01-16:
Ok not finding the settings, sorry, not trying to be thick here!

ghenry, 2022-01-16:
Settings on GitHub or Coverity?

revk, 2022-01-16:
Coverity - I have account settings. It has a thing asking me to add projects...

revk, 2022-01-16:
Ok I added solar system on there

revk, 2022-01-16:
Ah, Ok I have a token.

ghenry, 2022-01-16:
Cool. Are you going to figure it out using the example SentryPeer one?

revk, 2022-01-16:
Somewhat out of my depth with this one, to be honest. I don't even see a .github directory let alone workflow, so I assume I have to make them. Happy to take any help I can get!

ghenry, 2022-01-16:
Yeah, create that folder and file in your local git repo and then commit
etc.

It will then show up here:

https://github.com/revk/SolarSystem/actions

You'll need to read the yml file and update it on how to build your project
as if you're on a Ubuntu machine. Ubuntu containers come with these
installed:

https://github.com/actions/virtual-environments/blob/main/images/linux/Ubuntu2004-Readme.md

Also, other things you can do for reporting issues:

https://github.com/SentryPeer/SentryPeer/tree/main/.github/ISSUE_TEMPLATE

and the free analysis tools from Github:

https://github.com/SentryPeer/SentryPeer/tree/main/.github/workflows

https://github.com/SentryPeer/SentryPeer/blob/main/.github/workflows/codeql-analysis.yml

revk, 2022-01-16:
OK on debian... I may try and catch up with this later...

ghenry, 2022-01-16:
Feel free to give me repo access and add me to the coverity project
and I'll set it all up for you. Then revoke my project access.
Probably easiest :)

commit 3c36f9a5 (2022-01-16): Closes #184